nekr0z/linux-on-huawei-matebook-13-2019 #3: headphone jack  detection

ffftwo, 2019-04-14:
Try this, it works for me.

> cat /etc/modprobe.d/sound.conf 
options snd_hda_intel model=dell-headset-multi

nekr0z, 2019-04-14:
Yep, perfect. Adding to readme.

closed 2019-04-14 by nekr0z

shostakalex, 2021-07-06:
Hi there, thanks! Can I do something with my audio here? 
```

[dreamndi@dreamndi-linlap modprobe.d]$ sudo alsactl restore
alsa-lib ucm_subs.c:252:(rval_lookup_main) Lookups are supported in v4+ syntax
alsa-lib parser.c:242:(error_node) UCM is not supported for this HDA model (HDA Intel PCH at 0x9421c000 irq 144)
alsa-lib main.c:1402:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -6
[dreamndi@dreamndi-linlap modprobe.d]$ alsactl restore
alsactl: state_lock:129: file /var/lib/alsa/asound.state lock error: Permission denied
alsactl: load_state:1636: Cannot open /var/lib/alsa/asound.state for reading: Permission denied
alsa-lib ucm_subs.c:252:(rval_lookup_main) Lookups are supported in v4+ syntax
alsa-lib parser.c:242:(error_node) UCM is not supported for this HDA model (HDA Intel PCH at 0x9421c000 irq 144)
alsa-lib main.c:1402:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -6
Found hardware: "HDA-Intel" "Realtek ALC256" "HDA:10ec0256,19e5320d,00100002 HDA:8086280b,80860101,00100000" "0x19e5" "0x3e09"
Hardware is initialized using a generic method

```

frusanov, 2021-09-29:
On AMD version of MateBook 13 combo jack start work properly only after adding `options snd-hda-intel model=alc255-acer,dell-headset-multi` to my /etc/modprobe.d/sound.conf

nekr0z, 2021-09-29:
@frusanov Interesting! Care to make a pull request with this info?

frusanov, 2021-09-30:
@nekr0z yes, I can. Can you check does same config work on intel version? I think better add to instruction one universal config

frusanov, 2021-10-04:
A speedy recovery! 

I create PR where I determined info about amd version separately, but in case if on intel version same code also works it's needs to be corrected to one single recommendation

nekr0z, 2021-10-05:
I'm merging it, and will test the Intel model and revisit the topic as soon as feasible.

opr-sergey, 2022-02-03:
On of my matebook 14s, headphones doesnt work after adding options `snd-hda-intel model=alc255-acer,dell-headset-multi` to my `/etc/modprobe.d/sound.conf`

I use apple headphones into jack 3.5mm, but play sound ONLY dynamics, at what microphone work correctly

mil-ad, 2022-03-17:
@opr-sergey do the headphones work _before_ adding those options? 

there's a relevant lower-level [discussion](https://giters.com/thesofproject/linux/issues/3350) on sof

commit ccfb6132 (2019-04-14): add headphone jack fix (fixes #3)