nekr0z/linux-on-huawei-matebook-13-2019 #4: Mic led

aymanbagabas, 2019-04-15:
Hi,

After looking at the DSDT table, I've come to these methods SMLS and PAFF which seem to turn something on and off, and also called from the AMW0 device (WMI device) in SSDT5. Using a tool like [acpi_call](https://github.com/mkottman/acpi_call) could you provide the return values and observe any changes to your system using the following?
```
$ echo "\_SB.PCI0.LPCB.H_EC.ECCC 0x48 0xA0 Zero Zero" | sudo tee /proc/acpi/call
$ sudo cat /proc/acpi/call
$ echo "\_SB.PCI0.LPCB.H_EC.ECCC 0x48 0xA1 Zero Zero" | sudo tee /proc/acpi/call
$ sudo cat /proc/acpi/call
$ echo "\_SB.PCI0.LPCB.H_EC.ECCC 0x45 0xA0 Zero Zero" | sudo tee /proc/acpi/call
$ sudo cat /proc/acpi/call
$ echo "\_SB.PCI0.LPCB.H_EC.ECCC 0x45 0xA1 Zero Zero" | sudo tee /proc/acpi/call
$ sudo cat /proc/acpi/call
```

Thank you!

nekr0z, 2019-04-15:
0x48 is mic mute led, great catch! 0xA0 turns it on, 0xA1 turns it off (both return `0x0called`). However, I can't see it change any EC registers.
0x45 returns the same `0x0called` but I fail to see what changes on the system.

aymanbagabas, 2019-04-15:
This should also work too.

```
$ echo "\SMLS 0x00010000" | sudo tee /proc/acpi/call
$ echo "\SMLS 0x00000000" | sudo tee /proc/acpi/call
```

I've seen this method in your model MB13 2019, mine MBXP 2018, and MBD 2018. This is called from the WMI device in all of these models.

nekr0z, 2019-04-15:
And indeed it does!

aymanbagabas, 2019-04-17:
I will open up an issue in [Huawei-WMI](https://github.com/aymanbagabas/Huawei-WMI) regarding this one. Right now, and since we've found how to get some of the other features to work, I feel this project, Huawei-WMI, should evolve to not only keyboard driver but also an nat/acpi driver.
I'm closing this issue.

closed 2019-04-17 by aymanbagabas

nekr0z, 2019-04-18:
> I've seen this method in your model MB13 2019, mine MBXP 2018, and MBD 2018. This is called from the WMI device in all of these models.

Does this mean your patch should work for enabling mic led on MB13 as is?

aymanbagabas, 2019-04-18:
No, not right now. I'm working on updating the module to fix this issue and to implement the other features discussed.