nekr0z/linux-on-huawei-matebook-13-2019 #5: Suspension Wake Up

joaomaridalho, 2019-04-20:
Hi,

I'm running manjaro 18.04 Illyria KDE when i suspend using the lid the system wakes up the screen randomly and consumes a lot of battery while in this state.

Do you have any recommendation?

nekr0z, 2019-04-20:
> Do you have any recommendation?

I can't say I have noticed such behaviour on Debian. The first thing to do would be to have a good look at system logs to see if anything unusual happens there while supposedly suspended, and maybe understand the reason for those wakeups.

nekr0z, 2019-05-18:
OK, I think I got this one. The build quality of the laptop is not perfect, and aluminium body has some give when pressed hard against. This leads to the lid pressing the keys and waking the laptop up when pushed and pressed against things in your backpack.

I see only two workarounds: either hibernate or use hard cases and be gentle with your laptop.

joaomaridalho, 2019-05-18:
Hey, that makes a lot of sense after I read that the drivers you recommended were merged into 5.0 I installed the mainline kernel which solved the error log spam(touchpad) issue and also solved this issue but now that you mentioned I now can only wake up the laptop using the power button

nekr0z, 2019-05-19:
> I now can only wake up the laptop using the power button

Double check the power settings of your system. I have no problem waking the laptop from sleep with keyboard using kernel 5.0. Of course, if the laptop is hibernated, you can only wake it with power button, but regarding this issue it's the whole point (since the issue was actually waking up from random keypresses).

kian1991, 2019-06-03:
So after testing, I disabled the boot grubloader in BIOS and still...Todays Situation:

I was using and charging my MateBook then I just closed the lid and went away. Unfortunately when I checked my Bag it was superhot and the MateBook was still running...not responsive as it probably overheated. I am really desperate thinking about sending it in

kian1991, 2019-06-03:
I will now upgrade my BIOS to 1.15 and report back

kian1991, 2019-06-09:
@joaomaridalho Well after upgrader my BIOS and setting "waking from other" in BIOS to "disabled" I have no more problems regarding this issue :-)

PeterParfenov, 2019-08-28:
I also had this issue. Looking at `/proc/acpi/wakeup` I found out that XHC (lspci shows that this is usb 3.1 controller: `USB controller: Intel Corporation Cannon Point-LP USB 3.1 xHCI Controller`).
Disabling it in `/proc/acpi/wakeup` fixed the issue for me:
```
sudo -s
echo XHC > /proc/acpi/wakeup
```