nekr0z/linux-on-huawei-matebook-13-2019 #6: What do you think of Ubuntu support?

romariolopezc, 2019-05-01:


nekr0z, 2019-05-01:
What works on Debian should generally work in Ubuntu, there's nothing that much Debian-specific.

kian1991, 2019-05-27:
Well I have running Ubuntu 19.04 and it works lika charm (even touchpad - no pinch to zoom though)

Annoying Issue:

When I use GRUB as default Bootloader sometimes the notebook wont go to sleep (even on Windows 10) it gets hot in my bag an runs outta battery. 

Did you expierience something similar @nekr0z ? 

Cheers

nekr0z, 2019-05-28:
No, I have not experienced any issues with putting the laptop to sleep. However, there is a possibility of it waking up unexpectedly, more info on this matter in #5.

kian1991, 2019-05-29:
ah yeah looks loke the same issue thanks @nekr0z ill have a look into the logs as soon I have the opportunity which log should I have a look at?

nekr0z, 2019-05-29:
> which log should I have a look at?

I believe Ubuntu is using systemd for all the sleep-wakeup process, so it all should be in generaly system log (available via `journalctl`). I don't really know if there's a way to see which ACPI event provoked wakeup, but at least you can make sure that system went to sleep properly and then woke up from ACPI event (presumably, keyboard activity).