nekr0z/linux-on-huawei-matebook-13-2019 #8: What is the maximum battery life you had with matebook 13 in your installation?

codeitlikemiley, 2019-05-24:
I wanna confirm if im the only one experiencing a fast battery depletion in my huawei matebook 13
its like  2 and half hours then the batery runs out..
 
before in windows im getting 6-7 hrs ... 

most of the time im just surfing web, and coding, and compiling stuff...

i should be getting a decent amount of batterylife like before in windows..

If anyone have a good advice please share it

nekr0z, 2019-05-24:
I am consistently getting far beyond six hours. Earlier this week I gave a 4-hours lecture running my slides from the laptop, and then used it for some text printing work for a couple hours. I started that day on 89% battery and ended it at about 17%.

whome, 2019-05-29:
I do not have a Matebook 13, but I have the similar Matebook X Pro. Mine Matebook X Pro has an i7 CPU and a Nvidia GeForce MX150 GPU. I am running Linux Mint 19.1 Cinnamon.

I also experience short battery time but only when I use the Nvidia GPU. When I use the Nvidia GPU the temperature increases an the fan runs more or less all the time.

But when I switch to the Intel GPU does the batteries last much longer and the fan almost never starts.  It looks like the Nvidiea GPU needs a lot more power than the Intel GPU in Linux.

I do not know how your computers are equipped, if you have the Nvidia or the Intel version. Matebook 13 are available with both.

hg8, 2019-05-29:
> I also experience short battery time but only when I use the Nvidia GPU. When I use the Nvidia GPU the temperature increases an the fan runs more or less all the time.

> But when I switch to the Intel GPU does the batteries last much longer and the fan almost never starts. It looks like the Nvidiea GPU needs a lot more power than the Intel GPU in Linux.

I can confirm that this exact same behavior happen on my Matebook X Pro and my Matebook 13. The behavior is the very same in Linux and Windows thought.

codeitlikemiley, 2019-05-29:
Ive notice battery life is very low when NVIDIA mx 150 is being used, the model i have has 25 watts.
So the worst battery life , good thing ive installed bumblebee and bbswitch.
This Turn off the graphics card, Before My laptop is always hot even when opening a web browser
but now the battery life improves to 5 hrs with wifi On. 
I also did undervold like 100 Millivolts for the cpu and gpu and cache (whatever it does) .
10% is consumed for 35 mins at  this set up . so more or less ill be getting 5-6 Hrs of battery life

codeitlikemiley, 2019-05-29:
Ive also installed tlp which handle other stuff for powersaving...

PeterParfenov, 2019-08-27:
> I wanna confirm if im the only one experiencing a fast battery depletion in my huawei matebook 13
> its like 2 and half hours then the batery runs out..
>
> before in windows im getting 6-7 hrs ...
>
> most [...]

I have Matebook 13 i5-8265U/UHD Graphics 620 configuration and Fedora 30 on top of it: can confirm very poor battery life. Right now I've no idea why it's happening. Powertop shows an awful amount  of wake ups which basically drains battery. I've attached zipped log files (html and cvs) from powertop. Does anyone know what this might mean?
Nvidia isn't the case since I have only integrated intel card...

[powertop.zip](https://github.com/nekr0z/linux-on-huawei-matebook-13-2019/files/3543397/powertop.zip)

PeterParfenov, 2019-08-28:
Update: I had switched to X.org from Wayland, and strangely enough it did the trick: now my averaged time life without a recharge is about 4-6 hours. Also, I've noticed a significant heat up whenever I open X11 apps on Wayland, but there's no such issue on X.org.