nekr0z/linux-on-huawei-matebook-13-2019 #9: About HiDPI support

validatedev, 2019-08-15:
What can you do when you try to change DPI to 150%? Cannot find a way that works perfect.

Kpeamop, 2019-08-15:
xrandr

nekr0z, 2019-08-15:
I used KDE to set DPI. It's far from perfect, but i'm not too worried: it generally works for the software I use most, and KDE support for HiDPI is far from perfect anyway and they know it, so I expect it to become better with time.

Kpeamop, 2019-08-15:
Евгений, в консольке xrandr, по идее должно быть установлено. xrandr умеет и разрешения задавать и масштабируемость менять. Для сохранения установок xrandr'a, сохраните в корне домашней директории файл, уточните в гугле  название файла.

validatedev, 2019-08-16:
Yeah, I know these methods but xrandr causes screen tearing (and high CPU usage) for me and I really don't like KDE. If there's GNOME-wayland solution with doesn't blur the XWayland applications, this is the best solution for me.

ASM-bler, 2019-08-25:
If you are using Gnome, enable fractional scaling, worked for me.
X11:
gsettings set org.gnome.mutter experimental-features "['x11-randr-fractional-scaling']"

Wayland:
gsettings set org.gnome.mutter experimental-features "['scale-monitor-framebuffer']"

validatedev, 2019-08-25:
> If you are using Gnome, enable fractional scaling, worked for me.
> X11:
> gsettings set org.gnome.mutter experimental-features "['x11-randr-fractional-scaling']"
> 
> Wayland:
> gsettings set org.gnome.mutter experimental-features "['scale-monitor-framebuffer']"

X11 solution causes high CPU usage and screen tearing for me, the Wayland solution has compability problem that occurs on XWayland applications which is blurry fonts and elements. These problems annoys me so lot.

ASM-bler, 2019-08-25:
Yeah, I can confirm the blurry fonts problem with Wayland. That's why I stayed with X11.
Regarding the tearing, I could dimmish it substantially by creating a config file _10-intel.conf_ under _/etc/X11/xorg.conf.d/_ with the following content.
`
Section "Device"
   Identifier  "Intel Graphics"
   Driver      "intel"
   Option      "TearFree"    "true"
EndSection
`

validatedev, 2019-08-25:
> Yeah, I can confirm the blurry fonts problem with Wayland. That's why I stayed with X11.
> Regarding the tearing, I could dimmish it substantially by creating a config file _10-intel.conf_ under [...]

Thanks for the helping me!

validatedev, 2019-12-19:
I found a way that works without any hassle!
Just create _10-monitor.conf_ file under _/etc/X11/xorg.conf.d/_ with the following content:
`

`

ovsienkonikita, 2020-01-13:
There is fractional scaling in gnome>3.32, and its work fine for me.
For wayland:
```$ gsettings set org.gnome.mutter experimental-features "['scale-monitor-framebuffer']"``` 
For X11:
```$ gsettings set org.gnome.mutter experimental-features "['x11-randr-fractional-scaling']"```
For undo:
```$ gsettings reset org.gnome.mutter experimental-features```

validatedev, 2020-01-13:
> There is fractional scaling in gnome>3.32, and its work fine for me.
> For wayland:
> `$ gsettings set org.gnome.mutter experimental-features "['scale-monitor-framebuffer']"`
> For X11:
> `$ gsettings set org.gnome.mutter experimental-features "['x11-randr-fractional-scaling']"`
> 
> After executing this command you will be able to set scaling to 125, 150, .. etc %
> 
> For undo:
> `$ gsettings reset org.gnome.mutter experimental-features`

If you are using these settings
In Wayland: X11 apps will look blurry.
In X11: There is a screen tearing. Well, you can fix with `Option "TearFree" "true"`, but you’ll encounter at least 30% performance loss and unable to watch a video properly.

randomprofilename, 2020-08-12:
I highly recommend use custom resolutions with 3:2 aspect ratio.
I'm using 1600x1066 on wayland, to get this resolution just added 'video=eDP-1:1600x1066@60' in  /etc/default/grub GRUB_CMDLINE_LINUX_DEFAULT. Works way better than fractional scaling.
But double check that your display on eDP-1.