nekr0z/linux-on-huawei-matebook-13-2019 #10: Issue with Huawei WMI and TLP - no boost frequency

jkenda, 2019-08-25:
Ever since I'd installed Huawei WMI I've noticed that the CPU won't boost. Its maximum frequency stays where I've set "CPU_SCALING_MAX_FREQ_ON_BAT" in TLP. Even manually setting TLP mode to AC (sudo tlp set ac) does not help. The only way I can get it to boost is if I reboot with the power cable plugged in.
Has anybody else noticed this/knows how to fix it?

<i>OS: Ubuntu 19.04</i>

validatedev, 2019-08-25:
Wanna ask a question that is not related the yours topic. How can you manage the DPI (fractional scaling) for your machine?

jkenda, 2019-08-25:
You have to enable an experimental feature in Gnome:
`gsettings set org.gnome.mutter experimental-features "['scale-monitor-framebuffer']"`
Then you go to `settings -> devices -> displays` and select one of the scaling options (150% works best for me).

But I have to warn you that there is a reason the feature is experimental. Most of the time it works fine, but it is very glitchy when switching between fullscreen and non-fullscreen programs.

<b>Please no off-topic comments from now on.</b>

nekr0z, 2019-08-25:
Can't confirm the issue: mine happily boosts up to 3.8 GHz when allowed to (i.e. plugged in or `sudo tlp ac`).
Debian 10