nekr0z/linux-on-huawei-matebook-13-2019 #11: Is it possible to add fingerprint support?

GhostFantik, 2019-08-31:


nekr0z, 2019-08-31:
Is it possible to add it to this Readme? Yes, sure. If you have the instructions on how to make this fingerprint sensor work in Linux, you can make a pull request, or just post them here and I add them.

Where to find those instructions? I don't know (or else I would have added them). Is it even possible to make this sensor work on Linux? I don't know (or else I would have written about it).

GhostFantik, 2019-08-31:
OK! Thank you for answer

fa1segod, 2019-09-11:
There's recently a Linux version(deepin) of Matebook 14, which also has a fingerprint scanner. Maybe someone can port the driver for other Matebook devices.

GhostFantik, 2019-09-17:
Yeah. If I'm not mistaken, the version for matebook 13 will also be

aetbaev, 2019-09-30:
Must be great!

shostakalex, 2021-07-06:
Thank you for your work! Any movements of matebook 13 fingerprint here?