nekr0z/linux-on-huawei-matebook-13-2019 #12: batpro: iopl failed

melvinkcx, 2019-11-22:
I'm using Matebook 13 AMD version, get this when trying to run batpro:
[image]

My `dmidecode`:
```
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.
Table at 0xAE5DB000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: HUAWEI
	Version: 1.04
	Release Date: 09/10/2019
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 8192 kB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.4
	Firmware Revision: 1.4

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: HUAWEI
	Product Name: HN-WX9X
	Version: M1010
	Serial Number: -
	UUID: 20191006-acd5-6498-ef71-acd56498ef72
	Wake-up Type: Power Switch
	SKU Number: C500
	Family: MateBook

Handle 0x0002, DMI type 2, 16 bytes
Base Board Information
	Manufacturer: HUAWEI
	Product Name: HN-WX9X-PCB
	Version: M1010
	Serial Number: -
	Asset Tag: TBD
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 23 bytes
Chassis Information
	Manufacturer: HUAWEI
	Type: Notebook
	Lock: Not Present
	Version: M1010
	Serial Number: -
	Asset Tag: TBD
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Heng-W19BL

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
	Socket Designation: FP5
	Type: Central Processor
	Family: Zen
	Manufacturer: Advanced Micro Devices, Inc.
	ID: 81 0F 81 00 FF FB 8B 17
	Signature: Family 23, Model 24, Stepping 1
	Flags:
		FPU (Floating-point unit on-chip)
		VME (Virtual mode extension)
		DE (Debugging extension)
		PSE (Page size extension)
		TSC (Time stamp counter)
		MSR (Model specific registers)
		PAE (Physical address extension)
		MCE (Machine check exception)
		CX8 (CMPXCHG8 instruction supported)
		APIC (On-chip APIC hardware supported)
		SEP (Fast system call)
		MTRR (Memory type range registers)
		PGE (Page global enable)
		MCA (Machine check architecture)
		CMOV (Conditional move instruction supported)
		PAT (Page attribute table)
		PSE-36 (36-bit page size extension)
		CLFSH (CLFLUSH instruction supported)
		MMX (MMX technology supported)
		FXSR (FXSAVE and FXSTOR instructions supported)
		SSE (Streaming SIMD extensions)
		SSE2 (Streaming SIMD extensions 2)
		HTT (Multi-threading)
	Version: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx  
	Voltage: 1.2 V
	External Clock: 100 MHz
	Max Speed: 3700 MHz
	Current Speed: 2100 MHz
	Status: Populated, Enabled
	Upgrade: None
	L1 Cache Handle: 0x0005
	L2 Cache Handle: 0x0006
	L3 Cache Handle: 0x0007
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Part Number: Not Specified
	Core Count: 4
	Core Enabled: 4
	Thread Count: 8
	Characteristics:
		64-bit capable
		Multi-Core
		Hardware Thread
		Execute Protection
		Enhanced Virtualization
		Power/Performance Control

Handle 0x0005, DMI type 7, 27 bytes
Cache Information
	Socket Designation: L1 - Cache
	Configuration: Enabled, Not Socketed, Level 1
	Operational Mode: Write Back
	Location: Internal
	Installed Size: 384 kB
	Maximum Size: 384 kB
	Supported SRAM Types:
		Pipeline Burst
	Installed SRAM Type: Pipeline Burst
	Speed: 1 ns
	Error Correction Type: Multi-bit ECC
	System Type: Unified
	Associativity: 8-way Set-associative

Handle 0x0006, DMI type 7, 27 bytes
Cache Information
	Socket Designation: L2 - Cache
	Configuration: Enabled, Not Socketed, Level 2
	Operational Mode: Write Back
	Location: Internal
	Installed Size: 2048 kB
	Maximum Size: 2048 kB
	Supported SRAM Types:
		Pipeline Burst
	Installed SRAM Type: Pipeline Burst
	Speed: 1 ns
	Error Correction Type: Multi-bit ECC
	System Type: Unified
	Associativity: 8-way Set-associative

Handle 0x0007, DMI type 7, 27 bytes
Cache Information
	Socket Designation: L3 - Cache
	Configuration: Enabled, Not Socketed, Level 3
	Operational Mode: Write Back
	Location: Internal
	Installed Size: 4096 kB
	Maximum Size: 4096 kB
	Supported SRAM Types:
		Pipeline Burst
	Installed SRAM Type: Pipeline Burst
	Speed: 1 ns
	Error Correction Type: Multi-bit ECC
	System Type: Unified
	Associativity: 16-way Set-associative

Handle 0x0008, DMI type 9, 17 bytes
System Slot Information
	Designation: PCI-1
	Type: x1 PCI Express x1
	Current Usage: Available
	Length: Other
	ID: 2
	Characteristics:
		PME signal is supported
		Hot-plug devices are supported
	Bus Address: 0000:00:02.2

Handle 0x0009, DMI type 11, 5 bytes
OEM Strings
	String 1: $HUA001CN10000
	String 2: $HUA0833
	String 3: OemString3

Handle 0x000A, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 2
		zh|CN|unicode,0
		en|US|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

Handle 0x000B, DMI type 16, 23 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 8 GB
	Error Information Handle: Not Provided
	Number Of Devices: 2

Handle 0x000C, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 4096 MB
	Form Factor: SODIMM
	Set: None
	Locator: DIMM 0
	Bank Locator: P0 CHANNEL A
	Type: DDR4
	Type Detail: Synchronous Unbuffered (Unregistered)
	Speed: 2400 MT/s
	Manufacturer: Hynix
	Serial Number: 00000000
	Asset Tag: Not Specified
	Part Number: --VK    
	Rank: 1
	Configured Memory Speed: 2400 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V

Handle 0x000D, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 4096 MB
	Form Factor: SODIMM
	Set: None
	Locator: DIMM 0
	Bank Locator: P0 CHANNEL B
	Type: DDR4
	Type Detail: Synchronous Unbuffered (Unregistered)
	Speed: 2400 MT/s
	Manufacturer: Hynix
	Serial Number: 00000000
	Asset Tag: Not Specified
	Part Number: --VK    
	Rank: 1
	Configured Memory Speed: 2400 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V

Handle 0x000E, DMI type 19, 31 bytes
Memory Array Mapped Address
	Starting Address: 0x00000000000
	Ending Address: 0x001FFFFFFFF
	Range Size: 8 GB
	Physical Array Handle: 0x000B
	Partition Width: 2

Handle 0x000F, DMI type 20, 35 bytes
Memory Device Mapped Address
	Starting Address: 0x00000000000
	Ending Address: 0x000FFFFFFFF
	Range Size: 4 GB
	Physical Device Handle: 0x000C
	Memory Array Mapped Address Handle: 0x000E
	Partition Row Position: Unknown
	Interleave Position: Unknown
	Interleaved Data Depth: Unknown

Handle 0x0010, DMI type 20, 35 bytes
Memory Device Mapped Address
	Starting Address: 0x00000000000
	Ending Address: 0x000FFFFFFFF
	Range Size: 4 GB
	Physical Device Handle: 0x000D
	Memory Array Mapped Address Handle: 0x000E
	Partition Row Position: Unknown
	Interleave Position: Unknown
	Interleaved Data Depth: Unknown

Handle 0x0011, DMI type 21, 7 bytes
Built-in Pointing Device
	Type: Touch Pad
	Interface: Other
	Buttons: 2

Handle 0x0012, DMI type 22, 26 bytes
Portable Battery
	Location: in the back, on the bottom side
	Manufacturer: DYNAPACK
	Manufacture Date: 08/03/2019
	Serial Number: 0566
	Name: -
	Chemistry: Lithium Ion
	Design Capacity: 0 mWh
	Design Voltage: 11400 mV
	SBDS Version: Not Specified
	Maximum Error: 1%
	OEM-specific Information: 0x00000042

Handle 0x0013, DMI type 248, 12 bytes
OEM-specific Type
	Header and Data:
		F8 0C 13 00 01 4C E0 00 00 00 18 00

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table

```

My `lsmod | grep wmi`:
[image]

My `uname -a`:
```
Linux melvin-PC 5.3.0-18-generic #19-Ubuntu SMP Tue Oct 8 20:14:06 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

Any insights on this?

nekr0z, 2019-11-22:
First of all, you shouldn't use batpro and fnlock scripts if you can manage without them. These are dirty hacks, and readme explicitly states that you should use the the driver instead. Since you're using kernel 5.3, the driver is already there, so why would you even need the script?

melvinkcx, 2019-11-23:
How do I adjust the battery protection threshold with the default driver?

nekr0z, 2019-11-23:
Several options there:

- you could manually edit `/sys/devices/platform/huawei-wmi/charge_thresholds`,
- you could use [a command line/systemd tool](https://github.com/qu1x/huawei-wmi),
- or you could use [a GUI](https://github.com/nekr0z/matebook-applet) I've developed.

Mind you, all those options are referenced in README or on the pages it links to.

melvinkcx, 2019-11-23:
Thanks for the prompt reply, I did try all of the above, but to no avail. The directory `sys/devices/platform/huawei-wmi` simply doesn't exist in my case, any clue?
[image]

nekr0z, 2019-11-23:
Might be that you have older version (2.0) of the driver bundled with your kernel. [Grab](https://github.com/aymanbagabas/Huawei-WMI) a new one. As far as I remember, battery protection is governed since version 3.0.

melvinkcx, 2019-11-23:
@nekr0z  Thank you very much! I had to build and install the new driver from source, it is indeed working now

closed 2019-11-23 by melvinkcx

nekr0z, 2019-11-23:
You're welcome.