nekr0z/linux-on-huawei-matebook-13-2019 #15: Battery protection not working

RuSern, 2019-12-29:
Hi, I can't seem to get the battery protection to work.
I am using a Matebook 13, running on Ubuntu 18.04 initially. I followed the instructions to install the Huawei-WMI drivers using dkms tarball. I used the version 3.2 one since that version of Ubuntu had linux kernel 5.0. 
However, after installation I couldn't find the threshold parameter files for controlling the battery protection. Am I doing it right? 
-->"Battery protection can accessed from either /sys/class/power_supply/BAT0/charge_control_{start,end}_threshold or /sys/devices/platform/huawei-wmi/charge_control_thresholds"

Or did I misunderstand the instructions? 
I then tried to upgrade to ubuntu 19.10 to be able to use the latest version of the drivers, but it still doesnt work. 

Could somebody help me out?

nekr0z, 2019-12-29:
I see you have already raised an issue in the driver repository, that's exactly the right place.

RuSern, 2019-12-30:
Thanks, I have another question, do you know of anyone who has tried to install Deepin linux on their Matebooks? Does it come supporting all the features available on the matebook like the battery protection etc? Was thinking if Deepin Linux would come with all the drivers available.

nekr0z, 2019-12-30:
I have never touched Deepin, nor did anyone I know. But seeing as how Deepin's own wiki links to this repository for Matebook 13 support, I suppose they don't really do any more better that others :)