nekr0z/linux-on-huawei-matebook-13-2019 #17: CPU heating after kernel update 5.3

ilnytskyi, 2020-04-28:
Anyone faced a problem?

Updated my 
Matebook to Linux Mint 19.3 along with linux kernel 
4.15 --> 5.3

The CPU temp went crazy 
This is just after a few minutes of usage (fans are active)
Had to enable nvidia drivers (with them CPU temp seems to be more stable)

But the problem is that the laptop is doing nothing and it show  50+ CPU temp
Downgrade kernel or upgrade? :)

On Windows 10 everything works well, and no problems with CPU temp

```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
```
 inxi -Fxz ouput
```
System:
  Host: uaer-MACH-WX9 Kernel: 5.3.0-46-generic x86_64 bits: 64 compiler: gcc 
  v: 7.5.0 Desktop: Cinnamon 4.4.8 Distro: Linux Mint 19.3 Tricia 
  base: Ubuntu 18.04 bionic 
Machine:
  Type: Laptop System: HUAWEI product: MACH-WX9 v: M11 serial: <filter> 
  Mobo: HUAWEI model: MACH-WX9 v: M11 serial: <filter> UEFI: HUAWEI v: 1.28 
  date: 03/15/2019 
Battery:
  ID-1: BAT0 charge: 43.7 Wh condition: 54.7/56.3 Wh (97%) 
  model: DYNAPACK HB4593R1ECW status: Unknown 
CPU:
  Topology: Quad Core model: Intel Core i7-8550U bits: 64 type: MT MCP 
  arch: Kaby Lake rev: A L2 cache: 8192 KiB 
  flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 31999 
  Speed: 600 MHz min/max: 400/4000 MHz Core speeds (MHz): 1: 697 2: 692 
  3: 645 4: 697 5: 648 6: 683 7: 698 8: 648 
Graphics:
  Device-1: Intel UHD Graphics 620 vendor: Huawei driver: i915 v: kernel 
  bus ID: 00:02.0 
  Device-2: NVIDIA GP108M [GeForce MX150] vendor: Huawei driver: nvidia 
  v: 440.82 bus ID: 01:00.0 
  Display: x11 server: X.Org 1.19.6 driver: modesetting,nvidia 
  unloaded: fbdev,nouveau,vesa resolution: 3000x2000~60Hz 
  OpenGL: renderer: GeForce MX150/PCIe/SSE2 v: 4.6.0 NVIDIA 440.82 
  direct render: Yes 
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio vendor: Huawei 
  driver: snd_hda_intel v: kernel bus ID: 00:1f.3 
  Sound Server: ALSA v: k5.3.0-46-generic 
Network:
  Device-1: Intel Wireless 8265 / 8275 vendor: Huawei driver: iwlwifi 
  v: kernel port: 3000 bus ID: 3c:00.0 
  IF: wlp60s0 state: up mac: <filter> 
  IF-ID-1: br-0a055b1654f2 state: down mac: <filter> 
  IF-ID-2: br-11c267e078c7 state: down mac: <filter> 
  IF-ID-3: br-330bb42bde16 state: down mac: <filter> 
  IF-ID-4: br-65a8cd924b64 state: down mac: <filter> 
  IF-ID-5: br-7724f76da2c5 state: down mac: <filter> 
  IF-ID-6: br-7b1c50a5f03a state: down mac: <filter> 
  IF-ID-7: br-b041a28dd8e1 state: down mac: <filter> 
  IF-ID-8: br-c49d4eccbd8b state: down mac: <filter> 
  IF-ID-9: br-c9a5d0a0f688 state: down mac: <filter> 
  IF-ID-10: br-dfb155f1ec14 state: down mac: <filter> 
  IF-ID-11: br-fcb4b75ad4c4 state: down mac: <filter> 
  IF-ID-12: docker0 state: down mac: <filter> 
Drives:
  Local Storage: total: 709.83 GiB used: 141.18 GiB (19.9%) 
  ID-1: /dev/nvme0n1 vendor: LITE-ON model: CA3-8D512 size: 476.94 GiB 
  ID-2: /dev/sda type: USB vendor: Samsung model: Portable SSD T5 
  size: 232.89 GiB 
Partition:
  ID-1: / size: 228.73 GiB used: 141.15 GiB (61.7%) fs: ext4 dev: /dev/sda1 
Sensors:
  System Temperatures: cpu: 60.0 C mobo: N/A gpu: nvidia temp: 64 C 
  Fan Speeds (RPM): N/A 
Info:
  Processes: 288 Uptime: 29m Memory: 15.39 GiB used: 3.23 GiB (21.0%) 
  Init: systemd runlevel: 5 Compilers: gcc: 7.5.0 Shell: bash v: 4.4.20 
  inxi: 3.0.32
```

nekr0z, 2020-04-28:
I didn’t see anything like that on my machine on either 5.3 that I was running for a couple of months, or 5.4 that I’m running now. However, my laptop is <code>WRT-EX9</code> (unlike your <code>MACH-WX9</code>), with different CPU and no Nvidia graphics, so I can’t help you out much.

(Originally published at: https://evgenykuznetsov.org/reactions/2020/16/)

nekr0z, 2022-08-20:
I didn’t see anything like that on my machine on either 5.3 that I was running for a couple of months, or 5.4 that I’m running now. However, my laptop is <code>WRT-EX9</code> (unlike your <code>MACH-WX9</code>), with different CPU and no Nvidia graphics, so I can’t help you out much.

(Originally published at: https://evgenykuznetsov.org/en/reactions/2020/16/)