nekr0z/linux-on-huawei-matebook-13-2019 #22: Tested: Matebook 13 2020 with intel i5 10th generation cpu

sumutcan, 2020-07-24:
Hi all,
I just wanted to report that I was able to run Linux Mint 20 Cinnamon and Kubuntu 20.04 flawlessly. Every functionality in the matrix on the README file works except fingerprint reader of course. Fn keys are fully functional.

Kubuntu is especially smooth on 8gb hardware.

I also find [Slimbookbattery](https://launchpad.net/~slimbook/+archive/ubuntu/slimbook) tool particularly useful for creating energy profiles.

rexer01, 2021-05-28:
buenas tardes bro, tengo el mismo equipo, pero no he podido instalar ninguna versión de LINUX, podrías ayudarme con información al respecto por favor, te lo agradezco de antemano landa.absalon@gmail.com

Toorero, 2021-06-01:
Please open a new issue for that and provide some more details in English so other users with this or a similar issue can follow along too.

Please avoid thread hijacking in the future and speak English or at least translate it with Google Translator, DeepL or whatever since this is the common language spoken here.