nekr0z/linux-on-huawei-matebook-13-2019 #30: Matebook 13 2020 (intel GPU / i5-102100U) - Arch linux with sway

kar200, 2020-11-29:
This is to say a big thank you for your detailed guide and help with the matebook. Unfortunately there is no other area where I can write about this. 

My question is that I cannot set manually the batery charging levels even with root for some reason. I found a script (I cannot find it again) to set the levels at boot using systemd.

I was using first Manajaro KDE and fractional scaling as well as hardware video decode for chromium work out of the box.

Then onto arch with sway (new to both so a bit of a learning curve) and setting the scale to 1.5 in the sway settings works fine but only for wayland apps.

Now I am using Arch with sway without fractional scaling. This might not suit everyone (as I don't use a second monitor) but I managed to get everything working in wayland and xwayland with no blurriness.

Some gtk apps will scale fine with gsettings (i.e firefox)
`gsettings set org.gnome.desktop.interface text-scaling-factor 1.5`

Electron apps will scale as well by settings the xft.dpi in the ~/.Xdefaults file (I set mine to 144).

QT apps scale as well.

Obviously it's a mess to setup but the result is well worth it but again only when using the laptop screen. 

Anyway big thank you again for your work and it is amazing that I do not hear the fan go of once on Arch where in windows as soon as it starts it goes loud for quite some time before settling.

nekr0z, 2020-11-30:
> My question is that I cannot set manually the batery charging levels even with root for some reason.

[This discussion](https://github.com/nekr0z/matebook-applet/issues/22) may appear relevant for your case. Then again, it may not, but it's hard to say without knowing what exactly goes wrong and how exactly your system reacts when you try to change battery charging thresholds.

kar200, 2020-12-01:
> [This discussion](https://github.com/nekr0z/matebook-applet/issues/22) may appear relevant for your case. Then again, it may not, but it's hard to say without knowing what exactly goes wrong and how [...]

Yep that helped a lot. Thanks again.
I have been experimenting with different distros and installs and for some reason sometimes the "echo" command works and sometimes it does not.

Now it is working with the linux-zen kernel so I can just set it myself manually.

closed 2020-12-01 by kar200