nekr0z/linux-on-huawei-matebook-13-2019 #31: Huawei Mate Pro X 2020 and battery level control

jerabaul29, 2021-01-17:
I have a Huawei Mate Pro X 2020 (the one with i7 10510 U).

Do you expect that this script:

https://github.com/nekr0z/linux-on-huawei-matebook-13-2019/blob/master/batpro

Will work on it? Or is there a risk it will break stuff as the script is given for Matebook 13 according to the comments?

If it is expected to work on the MPX20, can you update the comments to make it clear to everybody? :)

nekr0z, 2021-01-17:
That script is a _very_ dirty workaround that nobody should even try to ever use. It has only been tested on pre-5.0 kernels, and only on model WRT-W19. I don't remove it from the repo for now, because current Debian Stable still has kernel 4.19, so the script might be of use to somebody, but I strongly advice everyone interested to install the kernel from Debian Backports (5.9 is already available there) and use the [proper driver](https://github.com/aymanbagabas/Huawei-WMI).

And in any case, the script is only tested for WRT-W19. For all I know, other models have different registry layouts, and trying to use this script on a very unsuitable model may do bad things to BIOS and even brick the laptop completely (maybe, nobody tried).

If you feel adventurous and want to take a shot at playing with your machine's registries, I'd suggest you read the discussion in #2 and get an understanding of what exactly the script does and how it can be adjusted to work on other models.

jerabaul29, 2021-01-17:
Ok, many thanks :)

The difficulty is that it is the first hit google provides - thanks for pointing to the more up to date location where to get this from!

nekr0z, 2021-01-17:
You may also want to use [the applet](https://github.com/nekr0z/matebook-applet) or at least read its readme, where it is all explained as extensively as I managed to. ;-)

jerabaul29, 2021-01-17:
Actually, after doing a bit of testing: installing Ubuntu 20.04, everything works out of the box. I do not need to install either this script or the code in the repo you suggest to be able to set battert thresholds: everything worked out of the box.

nekr0z, 2021-01-17:
As it should. The driver is integrated in the kernel since 5.3 (if I remember correctly), and Ubuntu is shipping what, 5.4 by default for 20.04?

jerabaul29, 2021-01-17:
5.8 :)

Amazing, I was just not aware all of this was working out of the box :) . Closing then!

closed 2021-01-17 by jerabaul29