nekr0z/linux-on-huawei-matebook-13-2019 #32: matebook 2020 amd, ppi 102/200

pooh2pooh, 2021-01-19:
Hi! The density of pixels per inch on my macbook 13 2020 amd 102. It is stated in the characteristics of 200. Free AMD GPU driver problem?
[image]

nekr0z, 2021-01-19:
Maybe, no idea. The value of 102 ppi you're getting is likely wrong, because (as it says just above it on your screenshot) at this resolution it would mean that you screen is 537 mm wide. I've never seen a 2020 Matebook 13 in person, but I seriously doubt they fit a 537 mm wide screen into a 286 mm wide laptop...

kar200, 2021-01-27:
For me it is shwoing 144 (changed the scaling factor). I have a matebook 2020 with intel and KDE but with a 1.5 scaling factor.
Not sure if that helps.

[image]

pooh2pooh, 2021-02-02:
Yes, Matebook 13 2020 AMD + KDE. 
[image]

closed 2021-06-30 by pooh2pooh